adafruit/Adafruit_NAU7802 #1: Suggestions to the load cell connectors at the PCB preview

ClemensGruber, 2020-04-14:
I have seen an announcement of the NAU7802 board at: https://blog.adafruit.com/2020/02/28/coming-soon-nau7802-icm20948-and-ds1841-stemmas-stemma-adafruit/ with this photo of the PCB. 

[image]

For the load cell connector you have the standard E+ E- A+ A- pins. Most load cells have shielded cable and you need a connection for this. So it would be nice to **extend the load cell connector for the shild** to E+ E- A+ A- shield (connected to ground). 

The NAU7802 is a 2 channel ADC, so it would be nice to have the 2nd channel also available on the break out. 

Many thanks @ladyada for your great products and code and all the best to NY from Berlin!

ladyada, 2020-04-14:
interesting idea! we may do so on next revision. thanx!

closed 2020-04-14 by ladyada

Meins321, 2020-04-28:
If you make such a connector please do it for 6 pin wheight cells as a 4 wire one can be used on 6 Wire connector with just two jumpers! (Bridged sense and power lines to the bridge)
the sense wire to the adc is split from the two power wires of the weightcell for closer measuring (wire resistance) that helps with drift and so on..

JohnHind, 2023-05-03:
@ladyada I'd just like to reenforce the desire to support the second channel. Why do I not just buy a second board? Well, sadly, this chip has no way to change the I2C address, so if I did that I'd have to use an I2C multiplexer board too! This design would be much better with a longer board and terminals for the second channel (and a ground terminal too as suggested).