adafruit/Adafruit_NAU7802 #2: CircuitPython driver for NAU7802

CedarGroveStudios, 2021-01-02:
I designed and tested a FeatherWing version of the NAU7802 ADC with a two-channel input. The code folder of the [repo](https://github.com/CedarGroveStudios/NAU7802_24-bit_ADC_FeatherWing) contains a CircuitPython driver partially based on the Adafruit NAU7802 Arduino driver and uses a similar UI. The docs folder contains a "pseudo readthedocs" [document](https://github.com/CedarGroveStudios/NAU7802_24-bit_ADC_FeatherWing/blob/main/docs/pseudo%20readthedocs%20cedargrove_nau7802.pdf) showing the UI.

When the Adafruit board is ready and if it helps, feel free to use any or all of the CircuitPython driver code as needed.

ladyada, 2021-01-02:
yay congrats! yes we did finisht hardware and then... march happened. oi! would you like to submit the NAU driver either to the community bundle OR we could add it to the adafruit repo and maintain it there (probably better idea so we dont need to involve ya in any bugfixes)

CedarGroveStudios, 2021-01-02:
I'd prefer to gift it to you "for the cause" so feel free to add it to your repo in any way you see fit. Attribution isn't needed, either. The benefits to me are 1) you and your staff will likely add value to it and make it work even better and 2) support and maintenance for your customer base will be easier and more responsive.

ladyada, 2021-01-02:
ha ok! will do :) i really outta get back to that breakout :/

CedarGroveStudios, 2021-01-02:
No reason to rush back to it on my account. I'm good to go.   ;)

ladyada, 2021-01-02:
ok gonna close this for now!

closed 2021-01-02 by ladyada

dhalbert, 2023-05-03:
Let's open a new issue to get this onto pypi.org, since that's the remaining step.