adafruit/Adafruit_NAU7802 #4: README.md - wrong link to product image

timov50, 2022-05-25:
I assume that the line below in the README.md
`<a href="https://www.adafruit.com/product/4226">[image]`

should be something like this:
`<a href="https://www.adafruit.com/product/4538">[image]`

evaherrada, 2022-06-17:
Fixed

ladyada, 2022-06-20:
https://github.com/adafruit/Adafruit_NAU7802/commit/5fb7c3ec87ae761de56b1efbd55a8e03ce715dec#commitcomment-76387205
@evaherrada

caternuson, 2022-06-20:
@evaherrada checkout #8 - was working another issue and was simple PR while i was here

caternuson, 2022-06-21:
Fixed with #8

closed 2022-06-21 by caternuson

evaherrada, 2022-06-21:
Ah sorry about that. Thanks for catching it!