adafruit/Adafruit_NAU7802 #5: Adafruit_NAU7802::begin() fails in revision register check

HamishB, 2022-06-16:
Hi,

I'm trying to get going with the Adafruit_NAU7802 example, but nau.begin() fails during the startup checks. A little debugging shows it is returning false here:

```
  /* Check for NAU7802 revision register (0x1F), low nibble should be 0xF. */
  Adafruit_I2CRegister rev_reg =
      Adafruit_I2CRegister(i2c_dev, NAU7802_REVISION_ID);

  if ((rev_reg.read() & 0xF) != 0xF) {
    return false;
  }
```

I found if I first upload SparkFun's NAU7802 example sketch, then without powering down the Arduino (a SAMD21 Feather) upload the Adafruit example sketch, then it works. Once I removed power it would get stuck again.

Commenting out the above check in the library meant it worked even after a power cycle.

thanks,
Hamish

ps- it would be really cool if future versions of the board had a couple pads on the bottom to bring out the second input channel's pins.

pps- it would also be really cool if the library supported outputting the chip's internal temperature sensor's reading, as strain gauges are highly temperature sensitive. (RFE, I know- will happen sooner if I help :)

HamishB, 2022-06-16:
Yes, the Adafruit breakout, shipped 10 days ago.

ladyada, 2022-06-17:
@HamishB could you ad a Serial.println() to determine what the value rev_reg.read()  is returning? cause thats odd - we use this library to test each board !?

HamishB, 2022-06-17:
rev_reg.read() prints out "0".

Hmmm. I tried with a different USB port, breadboard, M0 Feather Adalogger, jumper wires, and NAU7802 breakout board, and after a bit of swapping parts around it seems that the trouble is with that particular M0 Feather (batch "0822"), not the NAU.

Wheatstone bridge uses 120 ohm resistors (so ~ 25mA load @ 3V).  Voltages in & out of the NAU board look ok & stable.  Soldering is all good.  i2c SDA & SCL lines viewed with an oscilloscope are ever so slightly ragged but look well within spec -- I couldn't really see much difference there between the bad M0 Feather and the good one. I was having some trouble with the bad M0 board yesterday when it wasn't showing up as a USB port when plugged in & I had to press the reset button twice to trigger upload-mode.

thanks

ladyada, 2022-06-17:
freaky! well - would you be up for submitting a PR where those checklines are removed?

caternuson, 2022-06-17:
Might also be worth posting in the forums with photos of your setup since this sounds like it's specific to a particular Feather, and works as expected with a different Feather. If I'm reading the above comment correctly? Could be a marginal connection or other hardware issue?
https://forums.adafruit.com/

caternuson, 2022-06-20:
Recreated using QT PY M0. Here's a scope running the library example. Getting same `0` return as indicated above.
[image]

Using the [CircuitPython library](https://github.com/CedarGroveStudios/CircuitPython_NAU7802), it returns the expected value:
```python
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit QT Py M0 with samd21e18
>>> import board
>>> from cedargrove_nau7802 import NAU7802
>>> nau = NAU7802(board.I2C())
>>> nau.chip_revision
15
>>> 
```

The CircuitPython library resets and enables the board in its init:
https://github.com/CedarGroveStudios/CircuitPython_NAU7802/blob/6863bad290b1c3b3c435a0831185a2f3221271f3/cedargrove_nau7802.py#L112-L115

The Arduino library attempts to read the chip rev register *before* the reset and enable. Moving the check to *after* those calls gets it working:
[image]

HamishB, 2022-06-29:
Yeah I just had it happen again with a different SAMD21 Arduino and was starting to come to the same conclusion. I have good feelings that your patch will solve it.

A perhaps related oddness in the library's example might also now be fixed, change the 'if' to 'while' when the begin() test fails. Otherwise it was just continuing on even if there was an error (e.g. mismatched chip rev) which would also skip past any other more important error, and the warning that it failed would scroll out of view quite quickly.

```
--- nau7802_test.ino.ORIG	2020-03-03 01:49:16.000000000 +1300
+++ nau7802_test.ino	2022-06-29 14:56:16.462309255 +1200
@@ -5,8 +5,9 @@
 void setup() {
   Serial.begin(115200);
   Serial.println("NAU7802");
-  if (! nau.begin()) {
+  while (! nau.begin()) {
     Serial.println("Failed to find NAU7802");
+    delay(3000);
   }
   Serial.println("Found NAU7802");
 
```

thanks,
Hamish

HamishB, 2022-06-29:
ps- FYI in our tests the NAU7802 is far out performing the HX711 as far as signal:noise & calibration drift goes. So glad to have this working more reliably.

adavid7, 2023-01-29:
I'm seeing this same problem, of nau.begin() failing on that register check, with a Metro.

Also, shouldn't the initialization code in nau7802_test have a "while(true) ;" in the fail clause?

```
#include <Adafruit_NAU7802.h>

Adafruit_NAU7802 nau;

void setup() {
  Serial.begin(115200);
  Serial.println("NAU7802");
  if (! nau.begin()) {
    Serial.println("Failed to find NAU7802");
  }
  Serial.println("Found NAU7802");
```

caternuson, 2023-01-31:
Please update to the 1.0.2 release of the library when it becomes available in the Arduino Library Manager:
https://github.com/adafruit/Adafruit_NAU7802/releases/tag/1.0.2

caternuson, 2023-02-21:
Closing. Should be fixed in 1.0.2 release. Can reopen if not.

closed 2023-02-21 by caternuson

adavid7, 2023-02-22:
Worked for me, thanks!