adafruit/Adafruit_NAU7802 #6: Adafruit_NAU7802 will not compile in Arduino 1.8.12 (MaxOS X)

Maneki9Neko, 2022-06-18:
When I cut and paste the code Adafruit_NAU7802 into the Arduino IDE version 1.8.12 (MaxOS X) and run the 'verify' routine I receive the following error code:  **test_sketch_NAU7802:57:16: error: 'class Adafruit_NAU7802' has no member named 'calibrate'**

1.  Does anyone else have the experience that this code will not compile?

2)  Typically when I cut and paste provided code into the IDE, it compiles without difficulties.

3) I am a Newbie, and **if this is not the correct forum for this query, I apologize.**  I would be very grateful for any guidance on how to resolve this. or where to obtain code that will compile. 

Thanks

Robert

Here is the full error code:

Arduino: 1.8.12 (Mac OS X), Board: "Adafruit Feather 32u4"

/Users/robertchave/Documents/Arduino/test_sketch_NAU7802/test_sketch_NAU7802.ino: In function 'void setup()':
test_sketch_NAU7802:57:16: error: 'class Adafruit_NAU7802' has no member named 'calibrate'
   while (! nau.calibrate(NAU7802_CALMOD_INTERNAL)) {
                ^~~~~~~~~
test_sketch_NAU7802:57:26: error: 'NAU7802_CALMOD_INTERNAL' was not declared in this scope
   while (! nau.calibrate(NAU7802_CALMOD_INTERNAL)) {
                          ^~~~~~~~~~~~~~~~~~~~~~~
/Users/robertchave/Documents/Arduino/test_sketch_NAU7802/test_sketch_NAU7802.ino:57:26: note: suggested alternative: 'NAU7802_EXTERNAL'
   while (! nau.calibrate(NAU7802_CALMOD_INTERNAL)) {
                          ^~~~~~~~~~~~~~~~~~~~~~~
                          NAU7802_EXTERNAL
test_sketch_NAU7802:63:16: error: 'class Adafruit_NAU7802' has no member named 'calibrate'
   while (! nau.calibrate(NAU7802_CALMOD_OFFSET)) {
                ^~~~~~~~~
test_sketch_NAU7802:63:26: error: 'NAU7802_CALMOD_OFFSET' was not declared in this scope
   while (! nau.calibrate(NAU7802_CALMOD_OFFSET)) {
                          ^~~~~~~~~~~~~~~~~~~~~
/Users/robertchave/Documents/Arduino/test_sketch_NAU7802/test_sketch_NAU7802.ino:63:26: note: suggested alternative: 'NAU7802_RATE_320SPS'
   while (! nau.calibrate(NAU7802_CALMOD_OFFSET)) {
                          ^~~~~~~~~~~~~~~~~~~~~
                          NAU7802_RATE_320SPS
Multiple libraries were found for "Wire.h"
 Used: /Users/robertchave/Library/Arduino15/packages/adafruit/hardware/avr/1.4.13/libraries/Wire
 Not used: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire
exit status 1
'class Adafruit_NAU7802' has no member named 'calibrate'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ladyada, 2022-06-20:
something is amiss with your installation, the funciton is there
https://github.com/adafruit/Adafruit_NAU7802/blob/master/Adafruit_NAU7802.h#L94
you should not be cutting and pasting, install the library thru the IDE and run the example. see the guide for more info
https://learn.adafruit.com/adafruit-nau7802-24-bit-adc-stemma-qt-qwiic

closed 2022-06-20 by ladyada

Maneki9Neko, 2022-06-20:
I did install the library through the IDE. (Never cut and pasted). **This said, it is very helpful to know that the problem here is probably with the IDE and not with the library itself.** Thank you for your prompt response,.