adafruit/Adafruit_NAU7802 #10: Adafruit_NAU7802 connected to Arduino Nano 33 IOT

mhdalassi, 2023-02-25:
Hallo,

I am trying to load the example Code of Adafruit NAU7802 but when its completed the usb port disconnects immediately.
I dont know what the problem. But that also happened with the Arduino Portenta H7.
I am thankful for any suggestions.

caternuson, 2023-02-25:
This sounds more like a basic program loading issue rather than a library issue. Are you able to upload a basic Blink example sketch OK to these boards?

mhdalassi, 2023-02-25:
> 

Yes I tryed and it did work.

caternuson, 2023-02-25:
The sketch you are uploading for the NAU7802 is the example sketch from this library?

Need more information about what is happening here:
```
when its completed the usb port disconnects immediately
```
The sketch compiles and uploads without any errors? But then the COM port disappears and never comes back?

mhdalassi, 2023-02-25:
> 

Yes that is exactly what is happening

caternuson, 2023-02-25:
There's nothing special about this sketch or library that would explain that behavior. Have you been able to run other sketches that use `Serial` and see the output in the Arduino Serial Monitor?

mhdalassi, 2023-02-25:
Yas I did it more than once but this time when I open the serial Monitor the COM port disconnects.

caternuson, 2023-02-25:
What serial monitor did you open? You said it disappeared after uploading sketch?

mhdalassi, 2023-02-25:
I am using Arduino IDE 1.8.19 and after the Code has been complied I opend the serial Monitor and befor it opend the error appiers.

caternuson, 2023-02-25:
It's still confusing what is happening. You say this is happening exactly:
```
the COM port disappears and never comes back
```
But it sounds like there is a COM port showing after the sketch uploads?

mhdalassi, 2023-02-25:
Yes that what I had in mind. But I did it again and realised that It happend after I tried to open the serial Monitor.
So after uploading the COM port still there.

caternuson, 2023-02-25:
And then you open the Serial Monitor and see errors?

mhdalassi, 2023-02-25:
It didn't open and the error showed up

caternuson, 2023-02-26:
What error messages are you seeing?

caternuson, 2024-06-05:
Closing due to lack of response.

closed 2024-06-05 by caternuson