adafruit/Adafruit_NAU7802 #15: Ability to select channel 2 (AIN2N/AIN2P) of the ADC

hattesen, 2023-11-24:
The current implementation will only read analog channel 1 (AIN1N, AIN1P) of the [NAU7802](https://www.nuvoton.com/resource-files/NAU7802%20Data%20Sheet%20V1.7.pdf), while the ADC has two fully differential channels.

Channel 2 selected by setting bit 7 of REG0x02:CTRL2:
[image]

I realize that the Adafruit [NAU7802](http://www.adafruit.com/products/4538) breakout board only exposes channel 1 (`AIN1N`/`AIN1P`) as `A-`/`A+`, but this proposed change would allow more complete support of the [NAU7802 ADC](https://www.nuvoton.com/resource-files/NAU7802%20Data%20Sheet%20V1.7.pdf) with very little effort.

Two additional API methods would allow the current channel to be set/read:
```C++
setChannel(uint8_t channel); // set channel 1 or 2
uint8_t getChannel(); // returns 1 or 2
```

Let me know if you would like me to submit a pull-request implementing this proposed API extension.

hattesen, 2023-11-24:
I now realize that a 6 year pull request exists for this feature: #13 

```C++
bool setInput(NAU7802_PGAInput input);
NAU7802_PGAInput getInput(void);
```

caternuson, 2024-06-05:
As you've noted, the Adafruit breakout only uses a single channel. So this feature could be a general source of confusion and other issues. There's some further discussion here:
https://github.com/adafruit/Adafruit_NAU7802/pull/9

It looks like the Sparkfun library supports both channels:
https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/blob/e6b91f8259a26b6b05a52fdf2777b71f7357e237/src/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library.h#L198

abbbe, 2024-12-05:
> I now realize that a 6 year pull request exists for this feature: #13
> 
> ```c++
> bool setInput(NAU7802_PGAInput input);
> NAU7802_PGAInput getInput(void);
> ```

Can't help wondering why this feature was not merged in... Yes 2nd channel is pointless on this board, there is a filtering capacitor there (as recommended by the data sheet), but it would give access to temperature readings which is useful.

closed 2025-06-04 by ladyada